jkuusama/LitePlacer-DEV #2: Application freezes when no USB-TTL converter / TinyG is connected

mrandt, 2015-03-05:
When starting the application without hardware controller connected, it freezes completely on my machine (Win7 x64), bloats up memory usage and has to be killed via Task Manager.

From first investigation, issue seems to be located in `LitePlacer.SerialComm.Write` or further up the call hierarchy...

I believe when no device is connected, communication should time out and TinyG should automatically go to "disconnected" state on UI.

jkuusama, 2015-03-05:
> I believe when no device is connected, communication should time out and TinyG should automatically go to "disconnected" state on UI.

Of course, that is how it should behave – and does, on my Win8 machine. I’ll investigate.

 

Best regards,

 

Juha

mrandt, 2015-11-18:
This seems fixed with latest versions. If no TinyG connected, application gracefully goes to "disconnected" state. Suggest close.

closed 2015-11-18 by mrandt