jkuusama/LitePlacer-DEV #3: Home X crashes machine when position is already close to limit switch

mrandt, 2015-04-08:
If the machine is already positioned far left when executing "Home X" from "Basic Setup", the machine will crash into the leftmost barrier. It attempts to go beyond and will not stop unless TinyG is reset, although it is tripping the limit switch.

I experienced this behaviour when hitting "Home X" button several times in a row.

It seems to me that the - when homing X axis - machine first moves a few milimeters to the left before actually starting homing procedure and during this time limit switches are not handled correctly.

 The other axis do not have this issue.

mrandt, 2015-04-24:
The issue does still occur with latest software (LitePlacer_Install_20_4_2015b.exe).

To reproduce: 
1. Go to "Basic Setup" tab
2. Click "Home X" button
3. Wait for X axis to be homed
4. Now click "Home X" again

-> Machine tries to move left and crashes into frame repeatedly until TinyG is reset

I noticed that X-axis always moves to the left slightly before the homing cycle begins - only happens on X-axis, not on Y.

jkuusama, 2015-07-30:
Can't reproduce.

> X-axis always moves to the left slightly before the homing cycle begins - only happens on X-axis, not on Y.

Does not happen on my machine.

jkuusama, 2015-08-27:
I can't reproduce this. The homing cycle does behave strage if the homing speed is too fast and homing acceleration too slow to allow the machine to stop with the physical limits it has. Maybe this is relevant, maybe not. Anyway, I'm not closing this but I'm not going to do anything about it either until I find a way to reproduce the issue.

zrecommerce, 2015-10-27:
Can we confirm the limit switch detection option is enabled? Would someone mind providing screenshots of all settings? I'd like to see if I can recreate the issue.

jkuusama, 2015-10-27:
I think this happens when the homing parameters are not set "right" relative to machine physical properties and the amount it needs to travel back to clear the switch. I haven't figured out what exactly happens and what "right" means here.

DC37, 2016-02-10:
Juha's comment agrees with my observations - what I have seen is that the homing speed must be set "slow", and the homing acceleration should be set "fast". My numbers are 3200 for acceleration and 2500 for speed, though lower numbers also work for both of them.

jkuusama, 2017-03-14:
TinyG bug, that we can't fix. TinyG starts looking for switches only after the homing cycle is properly started, meaning start acceleration cycle is finished. Setting acceleration fast and speed slow is indeed the "solution", that allows the machine to get to homing speed even if the travel to switch is small.

closed 2018-09-07 by jkuusama