jkuusama/LitePlacer-DEV #6: Application window does not fit smaller screens - buttons hidden

mrandt, 2015-04-08:
I have a 19" TFT permanently attached to my LitePlacer frame. The display has a resolution of 1440x900 which I believe is reasonable for the display size.

Unfortunately the LitePlacer application window does not fit the screen and when maximizing, some buttons and labels at the bottom are hidden from view and out of reach.

I believe, with a bit of rearranging and smart use of the space available, the application could still be made to fit smaller screens. Otherwise a scroll bar would be better than just hiding controls "off screen".

I will attach a screenshot later to illustrate the issue.

mrandt, 2015-04-08:
See screenshot for illustration of the issue:

[image]

jkuusama, 2015-04-20:
> I believe, with a bit of rearranging and smart use of the space available...
For the Basic Setup screen, yes. A bigger issue is the Run Job screen, where it would be good to have the CAD data, the job data, the camera image and the buttons on the screen. The scroll bars are there now as a first aid, but I leave this open.

mrandt, 2015-11-18:
I am not sure what display and resolution everybody else is using, but I still have to scroll when using my 1440x900 display. 

It probably got worse with latest versions as those have more controls on the bottom section... :-(

mrandt, 2015-12-01:
I changed the layout to support 900px vertical resolution.

I have put the changes into a separate branch and created pull request:
https://github.com/jkuusama/LitePlacer-DEV/pull/38

jkuusama, 2015-12-29:
This has been out for a while now without bad comments. Closing.

closed 2015-12-29 by jkuusama