jkuusama/LitePlacer-DEV #7: Feature Request: Position confidence

mrandt, 2015-04-08:
I suggest to introduce some sort of position confidence management - could be very basic for now.

The application could keep an internal flag that stores if the current position is "valid". Flag should be enabled after the machine is homed and disabled again whenever TinyG is reset, connection is lost or motor power is disabled.

If software is not confident about the actual position, it should not allow any movements and request the user to home the machine first. 

This would greatly reduce the risk of "banging the machine into the wall".

For experts, we could still keep an option to override so they can still crash their machines ;-)

jkuusama, 2015-08-27:
For reset and lost connection we can do this. There is no message on power disable on the firmware that we are using. (Migrating to the latest TinyG firmware is an issue in itself...)

jkuusama, 2017-03-14:
I added our own timer to software that triggers when the TinyG motor power timeout triggers. You are prompter for homing and the home button is red, but I decided not to put guards for a non-homed machine. If you ignore the warnings, you can blame yourself for crashes. :-)

closed 2017-03-14 by jkuusama