jkuusama/LitePlacer-DEV #10: Feature request: Switch on / off LED light for upward facing camera

mrandt, 2015-04-23:
I have built quite a powerful led ring (monochromatic red at this point) for the upward camera. I would like to turn it off when not in use so it does not blind me... 

I think the TinyG SpinDir pin (logic level output) could be used to achieve this.

The LitePlacer software could pull this pin high whenever the upward camera is used, e.g. needle wobble calibration, component alignment etc.

I might start to work on this myself but still thought I should record it here.

jkuusama, 2015-04-23:
You would do this in SelectCamera() routine in MainForm.cs. Should be an easy addition.

 

Best regards

 

Juha

MaWa53, 2015-07-17:
I think the TinyG SpinDir pin (logic level output) could be used to achieve this.

I have implemented extra FET switches for both leds. 
But it is hard to set SpinDir with the unmodified TinyG firmware. 
It is combined with the spindle M3 M4 commands which - I don't know why - turns on the pump!
And after a TinyG reset it is toggled a couple of times.

Maybe the two output pins SS1 & 2 are better candidates. But I will have to take a closer look at the TinyG code to see if these pins are not already in use.

mrandt, 2015-07-17:
TinyG would be a much more useful piece of hardware if they had exposed some more analogue and digital pins - the MCU has plenty and some solder points would have been sufficient for most needs.

If we want LED control, pump control and maybe pressure sensor and a few other things, we quickly run out of options.

I am still considering building a add-on board which would feature a separate MCU and could connect to TinyG via SPI - more effort but also more versatile. That board could interface LED, control PWM or even RGB strips (thinking neo pixel or similar...), support one or multiple analogue pressure sensors and expose a few spare pins for things we come up with later. Will see if and when I find the time :grin:

jkuusama, 2015-07-30:
Not worth the effort in v1.

closed 2015-07-30 by jkuusama