jkuusama/LitePlacer-DEV #11: Feature request: Support for higher resolution cameras in LitePlacer software

mrandt, 2015-04-23:
I am experimenting with a custom 720p cam for uplooking vision and 1080p for downlooking vision (both modified computer webcams).

Resolution (pixel per mm) is obviously higher than with the pen cams from Juha's kit and also the noise is significantly lower with the CMOS chips I am using - which will hopefully allow higher accuracy for component orientation / detection and positioning.

However, software support is not 100% working yet. The camera image is scaled correctly on the tabs "Camera Setup" and "Run Job", I only see a smaller viewport (basically top-left corner of the full size) on the "Basic Setup" screen - which makes needle position calibration impossible.

My idea to extend LitePlacer for supporting higher resolution cams is the following:

1. During camera setup, it should be possible to

a) Set the desired hardware resolution which the chip should send to computer (e.g. 640x480, 1024x768, 1280×720, 1920×1080 pixels). I can specifiy the "size" parameter in VLC for example, so it should be possible. Using a lower resolution often results in less noise and also higher frame rate. Also, this setting would allow to adjust for aspect ratio (4:3 vs. widescreen picture).

b) Define a "viewport" to be used in program. This would basically mean "cropping" the image to a suitable size and aspect ratio. That way, we could use a smaller portion of the total image. This would be beneficial e.g. with lenses that distort the images at the borders and again to use cameras with an aspect ratio different than 4:3.

c) As a bonus, some basic corrections such as mirroring the image or turning it by 180° could be applied here. Could come in handy if the camera can only be mounted in one direction - but this is less critical.

2. The previously defined resolution, viewport and corresponding scaling shall be applied in all other dialogs of the program and also for automated computer vision (e.g. optical homing, needle wobble calibration).

Does this make sense?

jkuusama, 2015-04-23:
I have to get and play with better cameras. In the meanwhile, if you want to play with this yourself, it would be the SizeMode property of the various picture boxes in the MainForm.cs design view. But yes, supporting better cameras would be good.

mrandt, 2015-04-24:
I have played with different Behavior.SizeMode parameters for the various PictureBox objects.

I noticed that in the main version of LitePlacer this setting is not consistent. It is "CenterImage" on some screens and "Normal" on others - which explains the observed behavior.

In my own local fork of LitePlacer software I tried the following settings:

- Setting all to "CenterImage" at least gives me a centered viewport to calibrate the machine and get things working. However, this way a large part of the actual image is cropped and I loose the advantages of higher resolution. Also, the viewing angle gets much smaller so my uplooking camera cannot "see" larger chips end 2 end and on the downlooking all features seem to appear very large - which is not necessarily a bad thing though.

- Setting all to "Zoom" gives me the full view of the physical camera resolution. But in this case the aspect ratio becomes interesting. One of my cams has a physical resoltion of 1920x1080 which is 16:9 widescreen format. "Zoom" SizeMode scales the image while preserving aspect ration, so I get a picture on screen which has white bars at top and bottom. Also tiny details now are projected to small so I have to increase the downcam zoom factor a lot. 

Conclusion: I think that the optimal approach will be something similar to what I described in my initial issue record, a two step approach:

a) We need to first configure the parameters of the hardware driver to tell the camera which resolution we want. No point in using FullHD, I would rather run the same chip with 720p only but higher framerate and less noise.

b) We need to configure a virtual viewport - the area of the picutre to be used - parts outside this area will be cropped and discarded.

I have not gotten to the details of camera hardware handling yet but I will continue to look into this once I get my head around VisualStudio and MS C# language... ;-)

jkuusama, 2015-07-30:
It uses CenterImage now on all views, and this is as far as I want to go in ver1.

closed 2015-07-30 by jkuusama