jkuusama/LitePlacer-DEV #9: Possible to crash Z-axis when calibrating needle height

mrandt, 2015-04-23:
When calibrating needle height, Z-axis can be crashed into board.

I noticed this when calibrating needle height and accidentally jogged Z-axis down instead of up. This gets worse when one presses one of the modifiers, e.g. ALT key - in the best case, this will bend and break the needle - but might cause worse damage.

IMHO, it should only be possible to jog Z-axis up when calibrating needle height. Also, modifier keys should be disabled so only baby steps are possible.

As an additional safeguard, X, Y and A movement could be disabled while performing said calibration, so one does not accidentially "scratch" the PCB.

jkuusama, 2015-04-23:
Disabling the Alt key might be good, as well as allowing only Z movement. You do need to be able to jog down, as you usually jog up until you see the needle leave the surface; correct setting is a few steps down from that. But you are right, you don't want to go deeper than the process started with. In fact, you never need to go deeper than the table top surface, which is almost always 1.6mm down from PCB top surface.

jkuusama, 2016-02-15:
Added a check: Now you must manually confirm Z movement below PCB surface + 1.6mm. Writing this, I realize that I need to allow manually changing the guard value, some people have the PCB on standoffs.

jkuusama, 2017-03-14:
The manually adjustable check has been there for a while already...

closed 2017-03-14 by jkuusama