jkuusama/LitePlacer-DEV #22: Tapes: When next part is changed manually

Tsillen, 2015-10-30:
Tapes: When next part is changed manually changed back to one or whatever the next part x,y is not automatically changed.

I.e. I changed it to part 1 , it thinks next part is at part 17

mrandt, 2015-11-18:
Bug still exists in version 17_11_2015. 

If I manually change the number in "Next" column (Tape Positions List), Next X and Next Y should be updated.

Also, when executing "Set Hole 1" I would assume that Next X and Next Y should be updated.

mrandt, 2015-11-30:
Copy & pasted related issue from: http://liteplacer.com/phpBB/viewtopic.php?f=10&t=208#p1478

If you:
1) Move to first component's hole
2) Add component
3) Set component hole
is works fine.

BUT, if you:
1) Click Add component
2) Move to first component's hole
3) Set component's hole
then program (wrong) takes "Next" position from the last hole added (previous component) and when you set component's hole, "next" position isn't turned into component's hole, but stays the same.

mrandt, 2015-12-01:
I have fixed these issues in code and created a pull request for my changes:
https://github.com/jkuusama/LitePlacer-DEV/pull/37

Please check and merge to main branch.

commit af1b6ef6 (2015-12-09): Fixed #22 next coordinates and number not being updated

- enhanced event handlers and functions for reset tape, set hole 1, etc.
to update next part number and coordinates if needed
- added convenience function tapes.UpdateNextCoordinates
- made "next part no" column editable in tapes grid and added event
handlers to calculate and update next coordinates on edit of grid
closed 2015-12-09 by jkuusama