jkuusama/LitePlacer-DEV #23: Idea: Check with vision of part is successfully picked

Tsillen, 2015-10-30:
Idea: Check with vision of part is successfully picked.
pick part -> go with cam to component location -> check if there is still a component.
This would replace the reza vacuum sensor method.

mrandt, 2015-11-18:
I would prefer it the other way round and rather check if part was placed correctly.

1. Before placing a part, take a snapshot of measured location on PCB using down camera (should be pads with solder paste).
2. Attempt to pick part and place it.
3. After placement, take second snapshot of measured location on PCB and compare to picture from before. Should be fairly easy to detect a significant difference if component was placed. If pickup failed, the will be no part and machine should attempt again.

We could also check both: Check for empty pickup location to make sure part part was picked successfully as suggested AND check that it was placed. This takes slightly longer but is even more failsafe.

Just checking after pickup does not really help IMHO, as part may "fall off" during travel.

Once we had above funtionality, it could be enhanced to also check placement accuracy (component center on board matches expected center, rotation as expected).

jkuusama, 2022-09-09:
This would add robustness to the system. Imo, continuing is not the right thing to do, at least it should be user option: Alert, make a note and continue (needs placement log file), just try again. Not easy, and there are more beneficial issues to solve, so it is unlikely I'll tackle this.