jkuusama/LitePlacer-DEV #26: Bug: placing bottom layer components rotated 180 degrees

Tsillen, 2015-10-30:
When placing the bottom of a pcb Altium says they are rotated 180 degrees.

To combat this all rotations should be deducted by 180 degrees to get the correct orientation.

I think this is also the case in other cad packages.
File:
https://mega.nz/#!dJtXGC4T!pFb17wAJggP9GT5SZ-zh6j1TPb7rPpbTWloRCpBEZto

Device on bottom layer:
[image]

pnp file:
[image]

DC37, 2016-02-10:
I seem to get similar issues even on the top layer - sometimes, the machine insists on doing a 180-degree turn to pick up the part (even when I have the part orientation set to 180 on the tape) and then undoes it when it goes to place the part.

jkuusama, 2016-02-15:
Fixed. [http://www.liteplacer.com/double-side-data-handling/](url)

closed 2016-02-15 by jkuusama

jkuusama, 2016-02-16:
Is the result wrong or just that it does unnecessary rotations? If result is wrong, do you have a file and a setup where this is reproducible?

DC37, 2016-02-16:
The machine just does unnecessary rotations.

jkuusama, 2016-02-16:
I could make another issue for optimizing the movements. However, the rotation is not the limiting factor and there are much more beneficial issues to tackle; in honesty, I don't think that issue would ever get attention. This peculiarity can stay.

DC37, 2016-02-16:
Yup, no worries.