jkuusama/LitePlacer-DEV #27: Can't place components when there are no feducials?

Tsillen, 2015-10-30:
Can't place components when there are no fiducial

Due to the fiducial measurement being off one would want to place components without this compensation. But it seems not to be possible?

jkuusama, 2015-10-30:
The system needs to map nominal (CAD) coordinates to machine coordinates. Currently, this is done By fiducials, and you could use vias or mounting holes as fiducials as well. How?
- It would be rather difficukt to put board manually down accurately enough, but I guess it could be done. Are you suggesting adding a capability to do this?
- Some other mapping method?

Tsillen, 2015-10-30:
My pcb zero point is really accurate. Due to the boards being cnc routed this is also pretty accurate.
I used mounting holes as fiducials. The nominal locations were more accurate than the measured locations.

I would just implement it as an option:
Do you want to use fiducials or place as is?

hzeller, 2015-10-30:
What I did before (in some PCB processing software I wrote) was to use SMD pads of two components far away on the board to use as "fiducials".

So maybe if the software knows where the pads are, then it can pick two components and find the pads (possibly with assistence by the user 'is this really the right pad of D7' or something).

Tsillen, 2015-10-30:
Normally I have fiduacials but I kinda forgot to place them.
The method @hzeller proposed would also work.

jkuusama, 2015-10-30:
The latest commit has checkboxes for skipping both needle calibration (requested privately) and board measurement. I didn't have much time to check that before I had to go for a weekend trip with my family, but you can try it out.

Tsillen, 2015-10-30:
Ok thanks. I have exams this week so not much time.
A good reference for vision stuff is the book "Digital Image Processing". I'm currently doing my minor in this subject and will be able to contribute when I have more time.

jkuusama, 2016-02-15:
Using vias or other round pads works. This has to do for a while. I'm  cleaning up the repo and closing this issue, as I won't do anything for it at this point.

closed 2016-02-15 by jkuusama