jkuusama/LitePlacer-DEV #29: Feature request: Manually teach first and last part position using camera (not sprocket holes)

mrandt, 2015-11-18:
I would like a function to manually teach first and last part position.

There are situations in which sprocket holes are difficult to detect or not even available (e.g. custom trays) but components are still arranged in a repeating pattern / equally spaced.

Procedure: 
1. Jog camera over first part's center and store first part location
2. Jog to last part and also store location. 
3. Input number of available components.
4. Other configutation for part position (component orientation, pickup levels, etc.) should stay the same
4. Software calculates part positions based on equal spacing and calculates tape / tray direction and rotation.

This would rely on #28.

caprisonne, 2016-07-19:
I support with this proposal. It would be the fallback-function for the currently missing up-camera offset correction. It would be an important function to place ICs with very little distances between the pins.

jkuusama, 2019-09-04:
(Cleaning up old requests)
This is the "use coordinates directly" option in tape edit dialog. You don't give the number of coordinates, but the pitch of the tape; same result.

closed 2019-09-04 by jkuusama