jkuusama/LitePlacer-DEV #32: Bug: Mouse scroll wheel turns A axis

mrandt, 2015-11-18:
Funny bug: As my vertical screen resolution is too low, I often use the scroll wheel on the mouse to scroll main window.

When I do this, I noticed that A axis is turning at the same time. 

I suppose this is a bug, not a feature? ;-)

jkuusama, 2015-11-18:
It is a feature (A axis jog). The bug is not to see if you are trying to scroll the view or fine-tune A position.

mrandt, 2015-11-18:
Ok. It is not very obvious; I would personally have expected scroll wheel would control Z-axis if there was mouse control for machine at all :-)

jkuusama, 2015-11-18:
Until there is a way to reliably get all UI events before any of the Windows controls*, there will be no Z axis keyboard jogging, other than F keys. Far too easy to crash the needle. 

 

(*: There isn’t (boooo @MS!), some controls have privilege.)

jkuusama, 2015-11-19:
> I get that. F-keys work fine for me so no need to mess with other keyboard combinations.
>
> How can I turn off the mouse wheel spinning A axis? I cannot control my habbit of scrolling the main window with that wheel... ;-)
>
> On a side note: I also often used the numpad to enter values in text input fields and where suprised when my machine suddenly moved.
>
> Is it possible to not capture key press events if a input field has focus? Otherwise I would like an option to disable num pad jogging :-P

> Is it possible to not capture key press events if a input field has focus?

 

If there is, I don’t know how (unless doing if focus on this control or focus on that control or ..., which I’m not going to do).

 

> Otherwise I would like an option to disable num pad jogging :-P

This should be easy.

jkuusama, 2015-11-23:
Added options to disable both mouse and numpad jogging. Further, I found how to disable jogging if focus is on input field.

closed 2015-11-23 by jkuusama