jkuusama/LitePlacer-DEV #34: Feature request: Support up-camera component rotation / offset correction

mrandt, 2015-11-18:
Up-camera shall be used to determine center and rotation of a component currently held by the nozzle.

This could be achieved by detecting the edges and / or pins using computer vision.

Software should apply necessary corrections to accurately place the part.

This is a must have to place large parts, ICs in TQFP or QFN (no leads) packages and might pace the way towards BGA or fine pitched parts.

tanelainla, 2015-11-30:
I would vote for this one also. This would also avoid the ugly events when nozzle misses the part from reel and puts needle inside the paste. Validating that the component was actually picked up is my most wanted feature at the moment.

TheFlav, 2016-03-30:
I would really like to see the up camera check component rotation and correct any off-center issues.  We are having issues with TSOP and QFN.

petegit, 2016-07-19:
This is one of the basic features for a P&P machine. 

Sure, it needs time to implement it, but it should be on the roadmap.

caprisonne, 2016-07-19:
The up-camera-feature is essential for placing of ICs with very little distances between the pins. With the current software you have to jog the machine by hand to its position . This can't be done effectively if you want to produce small series with the liteplacer.

sp-apertus, 2016-12-22:
+1
this should get priority

KarlZeilhofer, 2017-06-29:
No improvements on this issue so far?  
I just had a look at this video: https://www.youtube.com/watch?v=PDmg4tf3Aio

caprisonne, 2017-06-29:
May I send you some Information in german about that thing❓

https://github.com/jkuusama/LitePlacer-DEV/issues/34#issuecomment-311872824

KarlZeilhofer, 2017-06-29:
of course you may. zeilhofer.co.at/kontakt

petegit, 2017-06-29:
Waiting for nearly two years, but nothing in sigth. May be switching over to OpenPnP software or better looking for an other P&P machine.

Phara0h, 2018-05-30:
+1 for this, please.