jkuusama/LitePlacer-DEV #40: Bug: Camera handling broken in latest release

mrandt, 2016-02-20:
I upgraded to 05_02_2016b and camera handling seems broken - at least with my cameras.

I tried Logitech C270 upcam, Andonstar and Ednet 1080p (webcam mod) downcam.

When I check "Keep Active" on either of the cameras, the cam preview always "flickers" between the two camera images. So I more or less see both pictures at the same time. Also, when attempting any operation that requires camera the program crashes with the following exception:

[image]

Full stack trace:
[keep_active.txt](https://github.com/jkuusama/LitePlacer-DEV/files/139115/keep_active.txt)

Without the "Keep Active" enabled and using fast switching ("Robust" checkbox unchecked), I also get an exception whenever I try to place parts:

[image]

Again full stack trace:
[place_robust_not_active.txt](https://github.com/jkuusama/LitePlacer-DEV/files/139116/place_robust_not_active.txt)

When enabling the "Robust" checkbox, it seems to work better - although I still managed get exceptions and "red X" camera preview during longer p&p jobs. Unfortunately I did not record the latter.

To me, the camera problems are a major inconvenience and thus I switched to Karl's fork for the time being.

dumbSail, 2016-02-23:
Having the same problem.

jkuusama, 2016-03-02:
I think/hope it is fixed now; at least, I haven't heard about problems on the latest release. ( <rant>Once more, this is one of those "works fine for me, not for customers" cases; I really should get an XP and a vista machine. </rant>)

closed 2016-03-02 by jkuusama