jkuusama/LitePlacer-DEV #46: Support for LED placement?

Kegsay, 2017-01-05:
Hello! Is LitePlacer able to correctly orientate 5mm LEDs (e.g. by recognising the outline using the built-in camera)? If not, is this due to fundamental limitations of the camera, or just because there isn't an accurate enough algorithm?

Thanks!

MaWa53, 2017-01-06:
The camera is not the issue. The algorhythm has been defined by jason von nieda in openpnp for the up looking camera which can also be used for the down cam. But the needed steps are not yet available in the liteplacer software. You need to adjust the angle addtionally to the x and y location. For 5mm LEDs the recognition works very well with white case material.

Kegsay, 2017-01-06:
Thanks for the quick reply! Are there any plans to support LED placement in the near future? Is there any way I can help?

closed 2019-07-03 by jkuusama