jkuusama/LitePlacer-DEV #53: Bug (5-28 release): Pressing stop while nozzle is placing an object puts the machine in a unrecoverable state.

Phara0h, 2018-06-01:
Well, first it does not actually stop the machine when pressed. The nozzle goes all the way down before stopping. After that is done, you can not bring back up the nozzle or do anything with the machine. It just says "can not move nozzle is down". When you click the home button nothing happens. The only way I got it to move up is if I hit the "manual change nozzle" button and that brings it up. However, you still can't move the machine. When you hit the physical reset button, the home still does not work. It will only home the z-axis and then stop. To get it to work again you have to reset the board, close the application, reopen it, connect to the board, re-home and remeasure before continuing the job. This can be quite frustrating when you are in the middle of a board job and you notice the machine failed to pick up a part so you need to stop it to correct the issue.

petegit, 2018-08-02:
I face the same problem!

jkuusama, 2018-09-07:
I believe today's release fixed this. There may be more cases where fuction return value is not honored, but I didn't find any. Closing this now, until somebody reports this again.

closed 2018-09-07 by jkuusama