jkuusama/LitePlacer-DEV #58: Software release, 07/09/2018: Manually assisted placement problem

petegit, 2018-09-16:
Manually assisted placement does not always work as expected!

If placing is started by selecting a part in the CAD data table and using the button _Place_ from CAD Data assisted placement does not work! After joging the part to final position hitting ENTER does not start the final placement **but moves the part with the nozzle still down!** 

Starting placing by selecting a part in the Job Data table and using the button _Place Selected_ from Job Data works as intended.

Attached two logs, one where placing was ok, one where it was nor ok. In both cases I directly pushed the ENTER button after the part was moved and positioned over the placement position. 

[place_jobdata_OK.txt](https://github.com/jkuusama/LitePlacer-DEV/files/2386775/place_jobdata_OK.txt)

[place_caddata_buggy.txt](https://github.com/jkuusama/LitePlacer-DEV/files/2386776/place_caddata_buggy.txt)

Looking into the logs at the line  _PutLoosePartDownAssisted_m: Now fine tune part position. If done press ENTER_ one can see the different behaviour. In the buggy log it looks like hitting ENTER is restarting the placment job!? 

jkuusama, I am not so deep into LitePlacer software and GCode, do you have any idea what went wrong?

Regards
Peter

jkuusama, 2018-09-17:
> do you have any idea what went wrong?

 

Not by these logs, but knowing how to get it misbehave allows me to reproduce the issue. And as always, once you can reliably reproduce it, it is usually easy to fix. I’ll investigate.

 

Best regards

 

Juha

jkuusama, 2019-08-13:
I think this is now fixed

jkuusama, 2022-09-07:
Done in 2019, forgot to close at the time.

closed 2022-09-07 by jkuusama