jkuusama/LitePlacer-DEV #61: Using arduino due and g2core

qiwalker, 2019-04-19:
Hi... 
Can i use arduino due and g2core with liteplacer software

Thanks
Rizki

jkuusama, 2019-04-19:
Hi,

Not directly. The software is open source; the needed modifications would not be big, if you handle the parameters setup elsewhere.

Best regards, Juha

jkuusama, 2019-07-03:
Support for g2core accepted for development

Symeth, 2020-10-26:
hi mr. Jkuusama

just for confirmation,
did liteplacer already support for g2Core ?

thanks

jkuusama, 2020-10-26:
Hi,
No, and i is unlikely it ever will, as there are no commercially available boards apparently, none in the future either. There is work towards Reprap firmaware boards (in the processing tables branch), but that it not ready yet.

jkuusama, 2022-07-26:
I will not do this. Reprap firmwareis much safer route.

closed 2022-07-26 by jkuusama