jkuusama/LitePlacer-DEV #74: Place fast miscounts quantity on partially placed board

jkuusama, 2019-07-03:


jkuusama, 2019-07-03:
"When i pause and resume "place fast" operation, it re-measures the tape, but counts the same quantity of parts as if no parts were placed yet. 
For example, if i placed 4 of 100 parts, it will search hole corresponding to part 105, which is most probably absent. 
And if i placed 90 of 100 parts, it will search part number 191, which is well outside the working area. 
But it clearly does not need this parts cos it will place parts with number no more than 100."

CRImier, 2019-07-03:
I've encountered this too, already looked into the code and decided that the code is not built in a way suitable for me to be able to make some sort of fix/workaround. IIRC the code just takes the part count number from the job table row, instead of re-counting by going through individual items in the part row and checking their "placed" status.

CRImier, 2019-07-03:
My workaround is, for now, placing a small piece of tape on the table, with a hole placed exactly where the PnP expects to see a hole ;-P won't work for 0402 as well, but for 0603/0805 it fools the PnP into thinking all is well, even if the components are not taken off the strip perfectly and are slightly misaligned, sure beats all the other solutions to this problem.