jkuusama/LitePlacer-DEV #78: Checkbox that makes "Reset tape" only reset the number (not Z)

CRImier, 2019-07-03:
Auto-probing Z (for both picking and placing) doesn't work for us, we hardcode Z offsets in our "Tapes" dialog. Is there any chance that you could add a checkbox that disables the behaviour of "Reset tape" button in the tape row ("Tapes" tab) that clears the Z picking and placing heights - we usually only need to reset the number. Usually, we just enter "1" into the "position", but sometimes we accidentally press "Reset tape" and it clears our Z settings, which is kind of frustrating and often results in the machine glitching while probing - then we need to reset the machine and restart the placement =)

jkuusama, 2019-08-06:
Reset tape  now only resets the count. There are other methods to reset Z values, but reloading a tape  (reset count only) wasn't easy

CRImier, 2019-08-07:
Oh, you did some changes? Thank you! Any chance they'll be available in one of the next releases?

jkuusama, 2019-08-07:
Yes, the very next one. If you want that now, you can compile it yourself, or if you are not set up for that, email me and I’ll do it for you,

jkuusama, 2019-08-13:
Relased now

jkuusama, 2019-08-13:
Implemented in release 14.08.

closed 2019-08-14 by CRImier