jkuusama/LitePlacer-DEV #103: Unhandled exception - 1st Fast Place in job from tape

lpuser13, 2019-08-31:
Exception details & screenshot are included below for reference.

- It seems this may occur with failure to recognise the tape hole. (happened several times after restarts etc)
- However, it should not cause an unhandled exception - which means I have to restart everything.
-It is happening on the first component & first hole on a new tape, using Fast Place (8mm tape 0805...I am using the same tape for all 0805 components for testing)
- It seems to go to the 2nd hole and gets that. Then it moves to the first hole and errors out.
- Hopefully I can get the video filter working better to avoid this, but it should be handled?

Details and screenshot follows:

> See the end of this message for details on invoking 
> just-in-time (JIT) debugging instead of this dialog box.
> 
> ************** Exception Text **************
> System.NullReferenceException: Object reference not set to an instance of an object.
>    at LitePlacer.FormMain.PickUpThis_m(Int32 TapeNumber)
>    at LitePlacer.FormMain.PickUpPartFast_m(Int32 TapeNum)
>    at LitePlacer.FormMain.PlacePart_m(Int32 CADdataRow, Int32 JobDataRow, Double X, Double Y, Double A, Boolean FirstInRow)
>    at LitePlacer.FormMain.PlaceComponent_m(String Component, Int32 GroupRow, Boolean FirstInRow)
>    at LitePlacer.FormMain.PlaceRow_m(Int32 RowNo)
>    at LitePlacer.FormMain.PlaceAll_button_Click(Object sender, EventArgs e)
>    at System.Windows.Forms.Control.OnClick(EventArgs e)
>    at System.Windows.Forms.Button.OnClick(EventArgs e)
>    at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
>    at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
>    at System.Windows.Forms.Control.WndProc(Message& m)
>    at System.Windows.Forms.ButtonBase.WndProc(Message& m)
>    at System.Windows.Forms.Button.WndProc(Message& m)
>    at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
>    at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
>    at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
> 
> 
> ************** Loaded Assemblies **************
> mscorlib
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.3416.0 built by: NET472REL1LAST_B
>     CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
> ----------------------------------------
> LitePlacer
>     Assembly Version: 1.0.7165.846
>     Win32 Version: 1.0.0.0
>     CodeBase: file:///C:/Users/Luke/Desktop/Arduino/LitePlacer/LitePlacerAug142019/LitePlacer.exe
> ----------------------------------------
> System.Windows.Forms
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.3324.0 built by: NET472REL1LAST_C
>     CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
> ----------------------------------------
> System
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.3416.0 built by: NET472REL1LAST_B
>     CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
> ----------------------------------------
> System.Drawing
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.3062.0 built by: NET472REL1
>     CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
> ----------------------------------------
> System.Configuration
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.3324.0 built by: NET472REL1LAST_C
>     CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
> ----------------------------------------
> System.Core
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.3429.0 built by: NET472REL1LAST_C
>     CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
> ----------------------------------------
> System.Xml
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.3062.0 built by: NET472REL1
>     CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
> ----------------------------------------
> Newtonsoft.Json
>     Assembly Version: 12.0.0.0
>     Win32 Version: 12.0.2.23222
>     CodeBase: file:///C:/Users/Luke/Desktop/Arduino/LitePlacer/LitePlacerAug142019/Newtonsoft.Json.DLL
> ----------------------------------------
> System.Numerics
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.3062.0 built by: NET472REL1
>     CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Numerics/v4.0_4.0.0.0__b77a5c561934e089/System.Numerics.dll
> ----------------------------------------
> System.Runtime.Serialization
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.3440.0 built by: NET472REL1LAST_B
>     CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Runtime.Serialization/v4.0_4.0.0.0__b77a5c561934e089/System.Runtime.Serialization.dll
> ----------------------------------------
> System.Data
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.3260.0 built by: NET472REL1LAST_C
>     CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_32/System.Data/v4.0_4.0.0.0__b77a5c561934e089/System.Data.dll
> ----------------------------------------
> AForge.Video.DirectShow
>     Assembly Version: 2.2.5.0
>     Win32 Version: 2.2.5.0
>     CodeBase: file:///C:/Users/Luke/Desktop/Arduino/LitePlacer/LitePlacerAug142019/AForge.Video.DirectShow.DLL
> ----------------------------------------
> AForge.Video
>     Assembly Version: 2.2.5.0
>     Win32 Version: 2.2.5.0
>     CodeBase: file:///C:/Users/Luke/Desktop/Arduino/LitePlacer/LitePlacerAug142019/AForge.Video.DLL
> ----------------------------------------
> Accessibility
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.3062.0 built by: NET472REL1
>     CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
> ----------------------------------------
> AForge.Imaging
>     Assembly Version: 2.2.5.0
>     Win32 Version: 2.2.5.0
>     CodeBase: file:///C:/Users/Luke/Desktop/Arduino/LitePlacer/LitePlacerAug142019/AForge.Imaging.DLL
> ----------------------------------------
> AForge
>     Assembly Version: 2.2.5.0
>     Win32 Version: 2.2.5.0
>     CodeBase: file:///C:/Users/Luke/Desktop/Arduino/LitePlacer/LitePlacerAug142019/AForge.DLL
> ----------------------------------------
> System.Web.Extensions
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.3429.0
>     CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Web.Extensions/v4.0_4.0.0.0__31bf3856ad364e35/System.Web.Extensions.dll
> ----------------------------------------
> System.Web
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.3429.0 built by: NET472REL1LAST_C
>     CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_32/System.Web/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Web.dll
> ----------------------------------------
> AForge.Math
>     Assembly Version: 2.2.5.0
>     Win32 Version: 2.2.5.0
>     CodeBase: file:///C:/Users/Luke/Desktop/Arduino/LitePlacer/LitePlacerAug142019/AForge.Math.DLL
> ----------------------------------------
> HomographyEstimation
>     Assembly Version: 1.0.0.0
>     Win32 Version: 1.0.0.0
>     CodeBase: file:///C:/Users/Luke/Desktop/Arduino/LitePlacer/LitePlacerAug142019/HomographyEstimation.DLL
> ----------------------------------------
> MathNet.Numerics
>     Assembly Version: 3.5.0.0
>     Win32 Version: 3.5.0.0
>     CodeBase: file:///C:/Users/Luke/Desktop/Arduino/LitePlacer/LitePlacerAug142019/MathNet.Numerics.DLL
> ----------------------------------------
> 
> ************** JIT Debugging **************
> To enable just-in-time (JIT) debugging, the .config file for this
> application or computer (machine.config) must have the
> jitDebugging value set in the system.windows.forms section.
> The application must also be compiled with debugging
> enabled.
> 
> For example:
> 
> <configuration>
>     <system.windows.forms jitDebugging="true" />
> </configuration>
> 
> When JIT debugging is enabled, any unhandled exception
> will be sent to the JIT debugger registered on the computer
> rather than be handled by this dialog box.

[image]

lpuser13, 2019-09-01:
..some more info (Showing version in use/latest) + log snippet at where the exception occurs.

> Application Start
> Version: 1.0.7165.846, build date: 08/14/2019 00:28:12
> Reading 
> 
> ....
> .....
> ......
> 
> 
> CNC_XY_m ok
> GoToFeatureLocation_m(), FindTolerance: 1.8, MoveTolerance: 0.1
> Downcamera stop: Stopped
> DownCamera start, moniker= @device:pnp:\\?\usb#vid_5149&pid_13d3&mi_00#6&32129a73&1&0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\global
> *** Camera started: 35
> Optical positioning, round 0, dX= -0.0683333333333333, dY= 0.0785833333333333, tries= 0
> Fast parameters:
> First X: 31.3116666666667, Y: 217.678583333333
> Last X: 31.3116666666667, Y: 217.678583333333
> Step X: 0, Y: 0
> PlaceComponent_m: Component: C1, Row:17
> PlacePart_m, Component: C1, CAD data row: 4
> Part position: Resistor100R, part #1: X= 29.3116666666667, Y= 221.178583333333
> Nozzle.Move_m(): X= 29.3116666666667, Y= 221.178583333333, A= 270
> CNC_XYA_m, x: 112.689741666667, y: 244.223408333333, a: 270
> ==> {"gc":"G0 X112.289741666667 Y243.823408333333 A270"}
> <== {"r":{},"f":[1,0,53,196]}
> <== {"sr":{"posy":217.600,"posa":0.000,"vel":0.26,"stat":5}}
> <== {"qr":31,"qi":1,"qo":0}
> <== {"sr":{"posx":32.196,"posy":217.864,"posa":2.723,"vel":2993.97}}
> <== {"sr":{"posx":39.768,"posy":220.318,"posa":27.990,"vel":13274.16}}
> <== {"sr":{"posx":56.087,"posy":225.608,"posa":82.448,"vel":20436.21}}
> <== {"sr":{"posx":75.249,"posy":231.818,"posa":146.393,"vel":20968.86}}
> <== {"sr":{"posx":94.610,"posy":238.093,"posa":211.002,"vel":18837.84}}
> <== {"sr":{"posx":107.936,"posy":242.412,"posa":255.470,"vel":9355.52}}
> <== {"sr":{"posx":112.123,"posy":243.769,"posa":269.442,"vel":1069.20}}
> <== {"sr":{"posx":112.290,"posy":243.823,"posa":270.000,"vel":0.00,"stat":3}}
> ReadyEvent stat
> <== {"qr":32,"qi":0,"qo":1}
> ==> {"gc":"G1 F150.0 X112.69 Y244.223408333333"}
> <== {"r":{},"f":[1,0,45,167]}
> <== {"sr":{"posx":112.290,"posy":243.823,"vel":0.76,"stat":5}}
> <== {"qr":31,"qi":1,"qo":0}
> <== {"sr":{"posx":112.578,"posy":244.111,"vel":150.00}}
> <== {"sr":{"posx":112.690,"posy":244.223,"vel":0.00,"stat":3}}
> ReadyEvent stat
> <== {"qr":32,"qi":0,"qo":1}
> Move Up

lpuser13, 2019-09-01:
progress....  :)

I double checked the Tape tab definitions for this tape & it seems the Width was set to 'custom' some how. It seems to work fine after I set it back to 8/4 width. So all is good again.

However, I suspect that this exception needs to be handled and maybe a check/prompt that there is a mismatch between the tape setting and maybe the job data?

Let me know if you want me to close this and you can add it to your list of TODOs.

jkuusama, 2019-09-03:
This should be handled, but I could not reproduce it. If you can make this happen again (try setting width to custom again?), please send me email. I'd like to get a zipped copy of your LitePlacer directory; I suspect a funny value in some data file, but I can't fix this if I can't make it happen. :(

lpuser13, 2019-09-03:
No Problem & thanks for the reply.

As it is handled, there is now no issue & I will assume my install/files was corrupted somehow. I remember having weird issues with my system that day. (some crashes/camera activation issues). 

Will close for now & if it arises again will send you the requested zip.

tnx

closed 2019-09-03 by lpuser13