jkuusama/LitePlacer-DEV #170: Tooltip text needs clarification for the "Load/set Current Position" buttons

Eriond, 2022-08-17:
The **Load Current Position** and **Clear Current Position** buttons in the bottom section of the window, have the same tooltip text. That makes the buttons' function unclear. I propose a text that is unambiguous, such as "Load the current nozzle position into the A-X boxes above, overwriting the content of all four boxes" and "Use the values in the A-X boxes as current nozzle position. Warning! This means that the nozzle will no longer be referenced to the home position"
Or something along those lines :)

jkuusama, 2022-08-17:
Thanks! The set position button tooltip now reads "Set current coordinates to X-A boxes content. Warning! Coordinates are no longer referred to homing results."

closed 2022-08-17 by jkuusama

AnalysIR, 2022-08-17:
Referenced is better than referred .....

Just saying Coordinates is also ambiguous to me.  Are you saying they are now referenced to machine (limit switches) vs after the homing sequence.

It might be better if you changed the colour in the xyza boxes to indicate if the machine is now operating using the 'homed' coords or the physical machine coords.

AnalysIR, 2022-08-17:
OK..
I seems you want to say that 0,0,0,0 (XYZA) is no longer the home origin based on the manual changes made, which is sort of obvious given the coords were changed. Maybe think about changing the home button to orange (vs red / grey) as a persistent indicator that coords have been manually changed by user (Until machine is Homed again).