jkuusama/LitePlacer-DEV #176: Sometimes, TinyG changes aam parameter by itself

jkuusama, 2022-11-14:
Sometimes, TinyG changes aam parameter by itself. I have no idea what causes this. So far, a few people have reported it and only one more than once. As a workaround, there is a check at connection (commit 1d826ad7e8a7212c83a336d44a2a0f4bc7cac9d0, 14.11.2022). If the value is not right, the correct value (1) is written. This is really a problem hiding workaround, not a fix!

ril3y, 2023-05-18:
https://github.com/search?q=repo%3Asynthetos%2FTinyG%20cm_set_am&type=code

This is odd.  cm_set_am is only called in this one place.  Wonder if there is some sort of memory corruption going on?

ril3y, 2023-05-18:
> Maybe. Or a timing issue associated with"set defaults" command, which
> writes all settings. In other words, lots of eeprom activity in a row. I
> suspect the latter, as even when this happens, a rewrite fixes it. Even
> before I hid this, nobody reported seeing the issue again.

Yes, the xmega needs small delays when writing to the EEPROM.  I don't
recall the exact amount but we had to do this in UI's when applying lots of
settings in a row.

jkuusama, 2023-05-18:
The software has that, with some margin. But wouldn’t be the first time that a part doesn’t always and in all cases work exactly as the data sheet says - and the workaround is already there, so this is on the open list more for principle than for need of fixing.